UTRA-ART/Caffeine #11: Connect the Navigation Stack

erickmu1, 2020-06-04:
**Description:**
- To enable autonomous navigation using nav goals, we use the Navigation Stack

**Goals:**
- [ ] Connect required topics to the navigation stack
- [ ] Run the navigation stack on our robot

**Acceptance Criteria:**
- [ ] Can autonomously navigate to any nav goal reasonably far
- [ ] No outstanding errors/warnings that can be resolved
- [ ] Configuration files are accurate to the specifics of Caffeine

**Notes:**
- It is not required to understand all the parameters of the Navigation stack at this point
- Primary goal is to get Caffeine navigating autonomously reasonably well without any in-depth parameter tuning

1999michael, 2020-07-21:
July 20th: Robot moves
Todo
 - Document parameters in a ROS wiki
 - Generate a map server by mapping with IGVC with gmapping
 - Look into localizing with AMCL. 
 - Test long distances

erickmu1, 2020-07-22:
> Generate a map server by mapping with IGVC with gmapping

Might as well do this in the RTAB Map issue, since it will come as a natural sub-task

> Look into localizing with AMCL

This should be done with RTAB, no reason to also add AMCL

> check whether we should use fixed odom or fixed base_link frame

What does this mean?

erickmu1, 2020-07-22:
@yvielcastillejos see #14 in terms of Documenting Nav Stack params. I started a wiki [here](https://github.com/UTRA-ART/Caffeine/wiki/Understanding-the-Navigation-Stack). You can either add to it or make a new one, I'll leave that up to you

1999michael, 2020-07-22:
> July 20th: Robot moves using rviz 2d nav goals using fixed odom frame
> 
> Todo
> 
> * Document parameters in a ROS wiki -> Yviel
> * Generate a map server by mapping with IGVC with gmapping -> Michael
> * Look into localizing with AMCL -> Later?
> * Test long distances -> Trudie
> * check whether we should use fixed odom or fixed base_link frame -> Michael

closed 2020-08-09 by erickmu1