UTRA-ART/Caffeine #14: Identify and Document Tunable Navigation Parameters

erickmu1, 2020-06-04:
**Description:**
- To perform a more in-depth tuning of the Navigation Stack and RTAB Map, we need to research which parameters are available and what they do
- Want to have an idea of how we will proceed to tune these packages

**Goals:**
- [ ] Identify tunable parameters in the Navigation Stack
- [ ] Identify tunable parameters in RTAB Map

**Acceptance Criteria:**
- [ ] Wiki identifying and explaining what parameters to tune for the Navigation Stack
- [ ] Wiki identifying and explaining what parameters to tune for RTAB Map

**Notes:**
- Questions to consider:
  - What is the parameter?
  - How does it affect navigation?
  - How can we determine a good value for it?
  - Difficulty in tuning?

erickmu1, 2020-06-27:
The section **Understanding the Navigation Stack** in our Wiki highlights the main components (ROS API) that pertain to the navigation stack. This should provide an overview on what the main components we can play around with are as well as their general functionality.

1999michael, 2021-01-07:
**Notes for documentation**
_Navigation Tolerance Tuning_
Location: nav/nav_stack/config/base_local_planner_params.yaml
- xy_goal_tolerance
- yaw_goal_tolerance

lukewarmtemp, 2022-01-22:
`global_costmap_params.yaml`: Tune width and height of the global cost map
`local_costmap_params.yaml`: Tune width and height of the local cost map