UTRA-ART/Caffeine #28: Succesfully navigate over ramps in IGVC world

1999michael, 2020-06-20:
**Description:**
- Successfully have caffeine navigate over ramps in the IGVC gazebo world

**Goals:**
- [ ] Successfully and reliably navigate over ramps in IGVC world

**Acceptance Criteria:**
- [ ] Navigate over ramps in IGVC world at various angles; straight & angled

**Notes:**
- This issue is raised as previously, we have attempted to navigate over the ramps, but have failed and slide back down. We suspect that it is related to the side wheels (which powers caffeine's motor actuation) being airborne as you climb over the ramp.
- Our current robot does not reflect caffeine's wheel suspension system that it has irl, which contributes to the idea of caffeine's side wheels being airborne

kajananchinniah, 2020-06-24:
Things we've tested:
- Driving the robot at 0.5 m/s, 1 m / s, 2 m/s in front of the ramp 
- Removing both casters and one caster and seeing performance 
- Spawning husky and having it go over the ramp 
- Setting the casters to continuous (with and without specifying an axis)
- Adjusting the height of the casters 
- Adding friction to the casters 

Conclusion we've come up with:
- Potentially a problem with the real life robot 
- Potentially a problem with how we simulated casters 

For now, this issue will be left alone and open.

1999michael, 2020-06-25:
potentially problem with simulated caster wheels or drive wheels?

1999michael, 2020-06-25:
Links:
 - https://github.com/trainman419/ackermann_vehicle-1/blob/master/ackermann_vehicle_description/urdf/em_3905.urdf.xacro
 - https://github.com/jbpassot/ackermann_vehicle?fbclid=IwAR2qRoa6eFRta-9FUW4SsmwslHJF7PASvGLbhRLHsrnvCuesiNXj2Gwuyqw
 - http://wiki.ros.org/ackermann_steering_controller?fbclid=IwAR1kOUr0Tme5lA2v3vWw98IwADnCadjFnGe9PhLSrecqmOAptPth_zUI1ak

Potential solution: Reference links above that have shock absorbers in their urdf files to see if this can work to simulate suspension system of drive wheel that caffeine has.

closed 2020-07-13 by 1999michael