UTRA-ART/Caffeine #46: Use robot_localization in 3D mode

erickmu1, 2020-08-09:
**Description:**
- Currently have an issue of Caffeine tilting forward/back which makes the Lidar see the ground and the Point Clouds not align with the ground plane in RViz. This is because we assume the robot is always flat by using 2D mode.
- The aim is to use `robot_localization`'s ekf nodes in 3D mode as to account for the tilt in reality

**Goals:**
- [ ] use ekf_local in 3D mode
- [ ] use ekf_global in 3D mode

**Acceptance Criteria:**
- [ ] Caffeine is seen tilted in RViz w/r to the map and odom frames
- [ ] Lidar scan is not flat when seen in RViz w/r to the map and odom frames
- [ ] Point cloud seems to be parallel to the ground w/r to the map and odom frames
- [ ] Navigating with keyboard teleop keeps drift to a minimum (can use caffeine/odom as ground truth)
- [ ] Can navigate with the nav_stack while using "3D odom"

**Notes:**
- The final test comes from when we actually connect the point cloud to the nav_stack and we don't filter out the laser scan by distance but by height.
- I believe a potential issue will be how odom/local is initialized (initial value), so lmk if you run into a problem with it (may be more apparent when we deal with the physical robot, as we can start ROS code during a tilted state - so we should try to account for this possibility in simulation)

1999michael, 2020-08-29:
**Current Progress**: Still trying to get fused local odometry to work reliably.  
**Observations**  
*Odom*
- If ZED camera faces out of map area, no visual features are observed and local odom shoots to +/- infinity along z-axis (Can fix by extending map with more grass)
- ZED camera occasionally resets, unsure of why. Once it resets, local odom goes bad, typically increasing/decreasing in z despite not being the actual case  

*Other*  
- Can remove bad specs from PointCloud2 rviz (point cloud) by increasing speckle_size in stereo_image_proc node in zed_emulation.launch
- When connecting ekf nodes with navigation stack, I believe there is a conflict for odom->base_link tf from ekf_local and move_base, generating error "TF extrapolation into the past". ekf_global with move_base conflicts are untested. Note that conflicting TF transforms using different time sources (Wall Time vs. Ros Time) will differ overtime. This will create lag in navigation stack, so error root cause + fix should be looked into

1999michael, 2020-10-10:
**Observations**

- Visual odom will get reset and stay at origin if it gets lost at any point (going up ramp seeing too much sky, looking outside map). This will make odom local/global go bad. This can partially (not deterministically) be prevented with lower velocities
- Being stuck. The encoders will think caffeine is moving, while visual odometry thinks caffeine is stationary. Overall local/global odom will think caffeine is moving.

closed 2021-01-07 by 1999michael