UTRA-ART/Caffeine #47: Connect PointClouds to the Navigation Stack

erickmu1, 2020-08-09:
**Description:**
- This issue should follow after #46 
- At this point, we should be able to un-comment the point_cloud observation source and use it for the navigation stack
- Moreover we should no longer need the hotfix of limiting the laser_scan by range, but rather by heigh

**Goals:**
- [ ] Connect point cloud to the nav_stack
- [ ] remove obstacle_range configuration and use an appropriate min_obstacle_height instead

**Acceptance Criteria:**
- [ ] Costmap is populated by both the Lidar and Stereo cam
- [ ] Can navigate autonomously with Lidar, Stereo cam, or both

**Notes:**
- Re-open #46 if there are persistent tilt issues

drinkerciku, 2021-02-16:
'min_obstacle_height' needs to be specified under the instance of each observation source in costmap_common_params.yaml file. the min_obstacle_height defaults to 0 in the master branch.

'max_obstacle_height' can be defined globally and overwritten locally (under specific observation sources instances) if needed.

drinkerciku, 2021-02-16:
obstacle_range cannot be removed as obstacles will not be registered at all --> empty costmap

drinkerciku, 2021-02-16:
The 'min_obstacle_height' works with reference to the visual odometry and if it is loss readings become inconsistent --> hard to navigate with the camera only. I also left some comments on Michael's issue in terms of visual odometry loss.

closed 2021-05-29 by 1999michael