UTRA-ART/Caffeine #53: Account for loss of visual odometry features in ekf localization

1999michael, 2020-10-17:
**Description:**
- When the zed camera has no visual features in sight, visual odometry is lost, and in ROS sets it to the origin. When the visual odometry is lost, the local/global odometry goes bad. Need to find a way such that either the visual odometry will never be lost, or that if it does, it won't have much of an effect on local/global odometry 

**Goals:**
We must ensure that either:
- [ ] Visual odometry is never lost (Visual odometry has enough visual features at all times, and will never reset to the origin)

    OR

- [ ] When visual odometry is lost, it can recover to the correct value and local/global odometry is still good

**Acceptance Criteria:**
- [ ] Good testing and justification to why we will never lost any visual odometry features

    OR

- [ ] Visual odometry is accurate before and after visual odometry being lost
- [ ] Visual odometry being lost does no throw off local/global odometry

**Notes:**
- This is one of two issues related to fixes to be implemented for ekf location (other issue is wheel slippage). You will need to remove situations where wheel slippage occurs when testing for this issue
- Note that the solution does not have to be restricted to changes directly related to ekf localization, as long as its reasonable

1999michael, 2021-01-25:
To do:
 - Extend map
 - Detect visual odometry reset

Potential workarounds for visual odometry reset
 - Use IMU + encoder data as replacement
 - drop visual odometry sensor source for odom local

What we still need to figure out
 - How to bring back visual odometry after reset / update visual odometry

drinkerciku, 2021-02-16:
Visual odometry is 'lost' while crossing the ramp and driving (the robot pitches back and forth when going forward and stopping). It is not always restored if it is lost while driving due to small discrepancies, but they accumulate with time and cause the robot to crash on obstacles. 

The odometry is restored when climbing the ramp successfully. However, I have not been able to test when going down the ramp because the robot ends up crashing on the ground (flipped on the side).

closed 2021-02-20 by 1999michael