UTRA-ART/Caffeine #55: Provide explanation to why the lidar does or does not register the ramp as an obstacle

1999michael, 2021-01-16:
**Description:**
- Through testing, we have gotten the navigation stack (move base) to go over ramp. This isn't expected as the lidar should register the ramp as a obstacle and try to go around it.

**Goals:**
- [ ] Identify whether the lidar registers the ramp as an obstacle
- [ ] Be able to explain why it does or does not register the ramp as an obstacle

**Acceptance Criteria:**
- [ ] Provide a good explanation for the observation

**Notes:**
- It would be good to change the spawn point to in front of the ramp instead of navigating everytime

MarcM0, 2021-02-20:
The laser scan detects the ramp as an obstacle (though it also detects the ground when the robot is leaning forwards). It seems that the point cloud can't see it and so its clearing it as an obstacle. As a result, the most of the ramp is not registered as an obstacle most of the time, with exception to the edges. This result also changes based on max and minheight of obstacles, and if the robot is leaning forward. If the point cloud is disabled, leaning forward detects the ramp, however, if the point cloud (and its clearing is on) the ramp is barely registered regardless. 
I changed the range of the laser sensor to 3.3 since that matches the real range of the sensor, and it picks up many more obstacles.
I am currently fine tuning max and min height to maximize seeing real obstacles and minimize seeing the ramp. Also, I'm setting the minheights individually as suggested in issue 47

MarcM0, 2021-02-20:
I have found some decent max and min values and I've saved them in costmap common param.yaml

MarcM0, 2021-03-13:
Update: with the new pull requests, the ramp is registered at an obstacle

closed 2021-03-20 by MarcM0