UTRA-ART/Caffeine #57: Add obstacles onto the local costmap

kajananchinniah, 2021-03-13:
**Description:**
- Given a coordinate array representing the location of obstacles, you should those obstacles onto the local costmap.

**Goals:**
- [ ] See obstacles on top of local costmap 

**Acceptance Criteria:**
- [ ] Does not affect global costmap 
- [ ] Does not overwrite the current local costmap
- [ ] Successfully adds the obstacles to the costmap 

**Notes:**
- Provide any additional notes/resources that might be helpful in the completion of this task.

kajananchinniah, 2021-04-03:
Update:
- Need to ensure that config files are up to date with master 
- Need to ensure that camera is only in global costmap and LIDAR is only in local costmap 
- Need a way to clear the obstacles after either every update or periodically (periodically is probably better)
- Need a method to linearly connect coordinate points together (see: https://en.wikipedia.org/wiki/Bresenham%27s_line_algorithm). Note, must ensure that we don't linearly connect points that are on different sides.

1999michael, 2021-08-07:
(seems to be useful)
https://answers.ros.org/question/348776/specifying-plugins-to-local_costmapyaml-causes-it-to-disappear/

1999michael, 2021-08-07:
Action Items (addition to kj's apr update)
 - Understand grid_layer vs. simple_layers cpp code does (http://wiki.ros.org/costmap_2d/Tutorials/Creating%20a%20New%20Layer)
 - (I think we should use grid_layers for saving history? need to be sure)
 - Interface with CV code (later). For now, use dummy array, that changes
    - From surface level looks, seems that we need to subscribe to a dummy topic and use callback function to process it

1999michael, 2021-09-12:
- Simple Layers: History is not kept
- Grid Layers: History is kept

1999michael, 2021-10-03:
Parent Class Documentation (look around for other inheritance classes)
- http://docs.ros.org/en/jade/api/costmap_2d/html/classcostmap__2d_1_1Layer.html

1999michael, 2021-10-03:
Main goal is to understand updateBounds and updateCosts functions, when they get called, and what they do

1999michael, 2021-11-06:
For testing reasons, we aim to work on global costmap, and see if we can build upon the old walls we added ourselves. 

This assume CV data is "stable" (changes can be made if assumption is wrong)

closed 2024-01-20 by Raindancer0912