UTRA-ART/Caffeine #63: P2

MarcM0, 2021-06-12:
**Description:**
- Address gps drift

**Goals:**
- [ Find the real values for the gps drift and put them in constants.xacro, find minimize the effects of gps drift in situations like CAFF-15]

**Acceptance Criteria:**
- [ Robot consistently navigates very close to correct waypoints (compared to when gps drift is turned off), waypoints do not glitch into or beyond walls] 

**Notes:**
- Provide any additional notes/resources that might be helpful in the completion of this task.

1999michael, 2021-06-12:
ideas:

- If we believe our global/local odom is more accurate, could we reverse calculate odom to GPS, and use that instead?

MarcM0, 2021-06-12:
> ideas:
> 
>     * If we believe our global/local odom is more accurate, could we reverse calculate odom to GPS, and use that instead?

Good idea. Where would we make that channge, move base or navigate waypoints?

1999michael, 2021-06-19:
I would actually publish a topic in the odom folder, and then reference that other topic from the navigate waypoints (if we so choose to)

1999michael, 2021-06-26:
What we should look into:

1. How does drift actually work (max/min, what direction, etc...) --> http://wiki.ros.org/hector_gazebo_plugins Looks to be the definition for the standard deviation
2. How does odometry/global values change with/without GPS drift
3. How does odometry/local values change with/without GPS drift
4. Overtime, we should find that the difference between odom value in GPS (GPS projected into odom frame) is close to odom global

How to convert GPS to odom frame for verification
1. Make a test node that converts GPS/fix to odom frame, similar to navigate_waypoints.py. This topic can be used to compare to odometry/global, odometry/local. NOTE: Can even do a topic that publishes the difference between the GPS/fix value in odom frame vs. the odometry global/local

How to do GPS -> odom
1. Make a python file and launch similar to navigate_waypoints.py
2. Python/CPP file should:
a. subscribe to GPS/fix
b. apply transform to GPS/fix to get the value in the Odom frame
c. Publish the transformed value
d. (Would be good) To make comparison easier, we can publish the delta of GPS/fix -> Odom with odometry/local & odometry/global instead of looking at 2 topics side by side

MarcM0, 2021-07-10:
Global seems to be much more off

local
deltaX= -0.00945254509127, deltaY = -0.0366929803624, deltaZ = 0.130747443928
deltaX= -0.00728453567912, deltaY = -0.0341913296042, deltaZ = 0.132291126843
deltaX= -0.00176391981054, deltaY = -0.0295726591105, deltaZ = 0.142932323065
deltaX= -0.00641377544494, deltaY = -0.0181510238029, deltaZ = 0.135232165053
deltaX= -0.00443098386487, deltaY = -0.0136316143235, deltaZ = 0.139571553013
deltaX= -0.0027260796278, deltaY = -0.0050328977039, deltaZ = 0.145287044658

global
deltaX= -0.0223258642276, deltaY = 0.135002561408, deltaZ = 0.232330038419
deltaX= -0.00729782131722, deltaY = 0.144283684989, deltaZ = 0.232834736193
deltaX= -0.0107647264243, deltaY = 0.142877473895, deltaZ = 0.236510794459
deltaX= -0.00473046484617, deltaY = 0.151286998808, deltaZ = 0.231771331052
deltaX= 0.00823983560928, deltaY = 0.153643468653, deltaZ = 0.231211182716

MarcM0, 2021-07-10:
these are all at spawn btw

MarcM0, 2021-07-10:
I ran a test where it went to the first few points of load waypoints (although we didn't get over the ramp and drift made the points innacurate)
Local: deltaX= 0.00521441586958, deltaY = -0.00133963894512, deltaZ = 0.00389438869625
Global: deltaX= -0.00560621930319, deltaY = -0.000333706758165, deltaZ = 0.00286965807019
Local: deltaX= 0.00368378434601, deltaY = 0.000245986885925, deltaZ = 0.00563379780318
Global: deltaX= -0.0029300469949, deltaY = 0.00299201016144, deltaZ = 0.0036024584565
Local: deltaX= 0.0254121298377, deltaY = 0.0053442682437, deltaZ = 0.00187527113442
Global: deltaX= 0.0226543386205, deltaY = 0.0064314700523, deltaZ = 0.00101386265335
Local: deltaX= 0.029670533207, deltaY = 0.0108266392404, deltaZ = 0.00100896983217
Global: deltaX= 0.0340278662935, deltaY = 0.0121905141055, deltaZ = -0.00424440127916
Local: deltaX= 0.0305540396574, deltaY = 0.00680005147102, deltaZ = 0.00318179716557
Global: deltaX= 0.0460509166452, deltaY = 0.00848830380818, deltaZ = 0.000145339819094
Local: deltaX= 0.0139634459731, deltaY = -0.00406374348426, deltaZ = 0.00575564662167
Global: deltaX= 0.0452111897595, deltaY = -0.00455514776981, deltaZ = 0.00215385984632
Local: deltaX= 0.0489402838557, deltaY = -0.00114843458803, deltaZ = 0.00519713497519
Global: deltaX= 0.0966613943199, deltaY = -0.000849325391162, deltaZ = 0.00173981088916
Local: deltaX= 0.0137494386797, deltaY = -0.0054668055073, deltaZ = 0.00483505039225
Global: deltaX= 0.100682473618, deltaY = -0.00506156410918, deltaZ = 0.00239647997834
Local: deltaX= 0.055557896037, deltaY = -0.0286066351219, deltaZ = 0.00404170054523
Global: deltaX= 0.136176232519, deltaY = -0.0285963273344, deltaZ = 0.000189916989112
Local: deltaX= 0.0721818552998, deltaY = -0.027210290357, deltaZ = 0.0039677627946
Global: deltaX= 0.170192310736, deltaY = -0.0216467368959, deltaZ = 0.00233069309154
Local: deltaX= 0.0850134339845, deltaY = -0.0200387317001, deltaZ = 0.00385820211182
Global: deltaX= 0.201948371307, deltaY = -0.0158873446022, deltaZ = 0.00268337935746
Local: deltaX= 0.0906788645535, deltaY = -0.0121963591259, deltaZ = 0.00270853490381
Global: deltaX= 0.220230933687, deltaY = -0.00604166795499, deltaZ = 0.00288348000827
Local: deltaX= 0.0947494370943, deltaY = -0.00923649949958, deltaZ = 0.00180167026887
Global: deltaX= 0.236097360196, deltaY = 1.36283544002e-05, deltaZ = 0.00239284217073
Local: deltaX= 0.104725768737, deltaY = -0.00485325047393, deltaZ = 0.00332003805846
Global: deltaX= 0.25978860786, deltaY = 0.0101679566616, deltaZ = 0.0013249894603
Local: deltaX= 0.115211565972, deltaY = -0.00875243474914, deltaZ = 0.00337352395667
Global: deltaX= 0.284863084539, deltaY = 0.00903830726086, deltaZ = 0.00394190482785
Local: deltaX= 0.133599955385, deltaY = -0.0102620050732, deltaZ = 0.00368156988363
Global: deltaX= 0.37699576774, deltaY = 0.0122899066817, deltaZ = 0.00594880347629
Local: deltaX= 0.159179324155, deltaY = -0.0251503362561, deltaZ = 0.00382696829772
Global: deltaX= 0.369280546153, deltaY = 0.00067421014955, deltaZ = 0.00492314735316
Local: deltaX= 0.122550904083, deltaY = -0.0245485040915, deltaZ = -0.000725244701396
Global: deltaX= 0.34147027457, deltaY = 0.00472312894025, deltaZ = 0.00024628958146
Local: deltaX= 0.206767704531, deltaY = -0.0260830885942, deltaZ = -0.00387671566735
Global: deltaX= 0.435454627472, deltaY = 0.00668712522019, deltaZ = -0.00260398904388
Local: deltaX= 0.180911928255, deltaY = -0.0218380207848, deltaZ = -0.00346065637495
Global: deltaX= 0.416518913682, deltaY = 0.012916163748, deltaZ = -0.00206605555983
Local: deltaX= 0.163055899726, deltaY = -0.0284387992992, deltaZ = -0.00580878611617
Global: deltaX= 0.400809320834, deltaY = 0.0100536907776, deltaZ = -0.00363470788037
Local: deltaX= 0.21741229032, deltaY = -0.0227015518024, deltaZ = -0.00637982104836
Global: deltaX= 0.423196893165, deltaY = 0.0180160065669, deltaZ = -0.00422827920889
Local: deltaX= 0.201077800526, deltaY = -0.0257346085115, deltaZ = -0.0066744884668
Global: deltaX= 0.4124089087, deltaY = 0.0192832825886, deltaZ = -0.00459007520258
Local: deltaX= 0.138072938788, deltaY = -0.0357445963964, deltaZ = -0.00702269704458
Global: deltaX= 0.39611263829, deltaY = 0.0130725494826, deltaZ = -0.0051175687278
Local: deltaX= 0.180413664331, deltaY = -0.0378437845543, deltaZ = -0.00981605775882
Global: deltaX= 0.399244162394, deltaY = 0.0164188317963, deltaZ = -0.00824739250047
Local: deltaX= 0.229584556321, deltaY = -0.027627474464, deltaZ = -0.0051423078078
Global: deltaX= 0.495505863797, deltaY = 0.0310189340222, deltaZ = -0.00277402705126
Local: deltaX= 0.244950920454, deltaY = -0.0466992736698, deltaZ = -0.00584677873504
Global: deltaX= 0.516751116667, deltaY = 0.0132064627702, deltaZ = -0.00396261005079
Local: deltaX= 0.153557685543, deltaY = -0.0418536066202, deltaZ = -0.00305130506168
Global: deltaX= 0.472704305218, deltaY = 0.0257436042005, deltaZ = -0.00507220173382
Local: deltaX= 0.161777137937, deltaY = -0.042209634957, deltaZ = -0.0100488454397
Global: deltaX= 0.4850684, deltaY = 0.0314695508807, deltaZ = -0.00835421846825
Local: deltaX= 0.228443927849, deltaY = -0.0485315971793, deltaZ = -0.0079524396777
Global: deltaX= 0.511810180348, deltaY = 0.0293873134151, deltaZ = -0.00623783019398
Local: deltaX= 0.205812104797, deltaY = -0.0420483192135, deltaZ = -0.00794273782357
Global: deltaX= 0.490936776402, deltaY = 0.0394766061636, deltaZ = -0.00626718614559
Local: deltaX= 0.206293320006, deltaY = -0.0423082308257, deltaZ = -0.00765805839539
[... cut ...]
Local: deltaX= 0.0504891237951, deltaY = -0.0534445144926, deltaZ = -0.0128366036016
Global: deltaX= 0.143640409128, deltaY = -0.141429606885, deltaZ = -0.00661084724675
Local: deltaX= 0.0459722429801, deltaY = -0.0668790103839, deltaZ = -0.0275639430336
Global: deltaX= 0.125510726871, deltaY = -0.149567850737, deltaZ = -0.0198182164899
Local: deltaX= 0.0434170610723, deltaY = -0.0744466495916, deltaZ = -0.0318293986088
Global: deltaX= 0.0913463011279, deltaY = -0.140607187236, deltaZ = -0.0221048791947

MarcM0, 2021-07-10:
I reran it a few times for 3 minutes and ploted the sqrt(deltaX^2+deltaY^2) (ignore the x axis, its just the number of datapoints)
(on the first run it got stuck at the end and on the last run the whole map drifted off near the end)
[image]
[image]
[image]

1999michael, 2021-07-24:
Observations:

- Local seems bumpier, probably due to IMU + smaller magnitude
- Why is IMU included in both global and local? I think IMU is a local sensor right?
- Does global converge (with/without IMU in global) when caffeine is stationary?

MarcM0, 2021-07-24:
Stationary:
[image]

MarcM0, 2021-07-24:
I turned off the IMU in global:

stationary:
[image]

moving(this was one of those runs where the gps waypoints happened to align correctly):
[image]

MarcM0, 2021-08-28:
We have been looking into manually setting values for the covariance matrices to have certain values converge slower (Like X and Y). You can change these settings in odom.yaml . The reason we do this is as you can see in the graphs above, some of our sensors seem to be having spikes of bad data, which we don't want to affect our position estimate above

1999michael, 2021-10-16:
Note: Details about non-sensor specific covariances in file Caffeine/odom/config/odom.yaml