UTRA-ART/Caffeine #80: P0 - Navigate without Human Intervention

Anthonyjlem, 2022-02-12:
**Description:**
- Currently for navigation, we manually set waypoints on Rviz, but we can't do this for the actual competition. The goal for this task is to figure out how to navigate without human intervention.

**Goals:**
- [ ] Caffeine navigates up to no man's land without pressing any buttons on Rviz on the old map

**Notes:**
- This task is blocked partially by ramps on the old map
- Suggestions:
- Do a long distance navigation to the waypoint (may not work)
- Set up small waypoints (don't have to get to them) along the way to the main waypoint
- Could hard code - have to change per map
- Check Google and other ROS packages

Anthonyjlem, 2022-04-30:
The rover currently tries to use a global planner to navigate from the start to the first waypoint at the start of no man's land, without setting any intermediate waypoints. The known issue blocking this from working is lidar readings are not being added to the costmaps, so the rover will drive into the walls.

1999michael, 2022-05-22:
No issues associated with odometry, costmaps, nor load_waypoints (points in json may be slightly off at worst). Only task left really is to tune a good planner. Currently known tunable parameters I see are:

(In the nav/nav_stack/config/ folder)
- TrajectoryPlannerROS parameters--velocities, cost function, or maybe a diff planner if they exist.(base_local_planner_params.yaml)
- Local / Global inflation radius (local/global_costmap_params.yaml)